pysco68/waveshare-hid #1: Question on "Finger" events

IoTGirl, 2016-10-28:
**From and IoT Core Team member:** Do I have correct understanding that your firmware does not send HID report for finger up event, only for finger down case ?

As I understand HID touch specification, each change of state must send a report.
I.e. the logic is as below

// byte[1] = state

If finger was up and now is down then
bState=FINGER_DOWN; // 1
Else if finger was down and is down then
bState=FINGER_DOWN_MOVED; // 3
Else
bState=FINGER_UP; // 0

Do you agree ?

closed 2016-10-31 by pysco68

IoTGirl, 2016-10-31:
Hi pysco68,

Were you able to fix your firmware? As this issue is now closed I am assuming you no longer need an answer to the question you posed above regarding time frame?

Sincerely,
IoTGirl

pysco68, 2016-10-31:
Hello IoTGirl,

The firmware is fixed, I'm still waiting for 24h to pass to see if I didn't introduce some bug (device will be powered continuously during that time and I will check on random basis if it is still responding as expected), but for now it's working okay.

In fact the touch response got better than with the original firmware, so I'm pretty pleased with the result.

Thanks for keeping in touch!

IoTGirl, 2016-10-31:
Hi pysco68,

Excellent! Better response time is even better news! I have let a few folks know about your alternate firmware and already have one reply:

_The waveshare display is pretty much the same as the Eleduino version…also you can burn this to the STM with VisualGDB on a PC, anxious to give it a try…_

Hopefully you will get some more feedback soon!

Sincerely,
IoTGirl

mastertychy, 2016-10-31:
Hello,
I'm glad that someone makes its working with Windows10! 
Unfortunatelly i'm not as smart to get rid with compiling whole firmware... i'm just beginner with stm32.
I'm using eclipse on windows and have StLink for programming/debugging.
Can you make and publish /send me already compiled file please (.elf file)?
In the time i'll try to make it compiles in linux cause it's too messy for beginner to edit compiler scripts ;)

commit 197799b3 (2016-10-31): Implemented sending HID touch up report after last "in move" report for any given finger

Fixes #1