pysco68/waveshare-hid #2: Add binary release

pysco68, 2016-11-01:
> NOTE: Issue submitted by @mastertychy on issue no. 1 / moved here as separate topic

Hello,
I'm glad that someone makes its working with Windows10!
Unfortunatelly i'm not as smart to get rid with compiling whole firmware... i'm just beginner with stm32.
I'm using eclipse on windows and have StLink for programming/debugging.
Can you make and publish /send me already compiled file please (.elf file)?
In the time i'll try to make it compiles in linux cause it's too messy for beginner to edit compiler scripts ;)

pysco68, 2016-11-01:
@mastertychy : let's discuss that here, in order to not disturb IoTGirl too much ^^ she's probably busy doing other important Windows 10 IoT stuff!

Regarding your request, I'm a bit reluctant to just provide plain binaries. I fear that people will just grab them and try flashing on other devices. 

Did you manage to get the repo compiled?

mastertychy, 2016-11-01:
Hi,

Actually, not ;). I was too lazy and use ubuntu repo arm-none-eabi-gcc... as i expected it won work :). I'll do it today with oryginal *tar.bz2.
About this LCD: I was using it with windows 8/8.1 for CarPC and it was working almost ok. LMB was working, RMB wasn't :). Touch behave more like single clicks when sliding etc. 
Now I "upgrade" CarPC (I'm using now intel compute stick compilant hardware) to windows 10 AU.
Sliding works great! RMB as well but... LPM isn't working almost at all.
Can you tell me if all functions are ok with yours firmware?

pysco68, 2016-11-01:
Hi,

my dev machine is running under Ubuntu-Mate 16.10 and I didn't use a custom built toolchain, just the plain Ubuntu one. If you got some warnings about some `2-byte wchar_t` conversion, you may just ignore them. I'd really like to get the self-compiling process to work, so it'll be great if you can help me a bit to figure out what doesn't work on your side (I hate the "it works on my machine" paradigm ^^)

Regarding your quesitons; both left and right button clicks work perfectly well, as do all other multi touch gestures under Windows 10.

mastertychy, 2016-11-01:
OK, i'm done. It was easy with non-repo arm-none-eabi-gcc ;). I've got main.elf file 320,2kB.
Now it's time to get StLink working on ubu (or i'll flash it under windows) i'll do this at the evening.

About my error with ubuntu repo - i'm working on 14.04LTS and error was with "--specs=nano.specs"... compiler was thinking it is a file "nano.specs" and it said it is missing :).

pysco68, 2016-11-01:
Okay,

my main.elf is exactly 317220 bytes long.
If you have an un-modified StLink you can use stlink tools from here: https://github.com/texane/stlink

This is a pretty good tutorial on how to install it: https://startingelectronics.org/tutorials/STM32-microcontrollers/programming-STM32-flash-in-Linux/

Otherwise flashing with the Windows ST-Link utility isn't much of an issue. You might need to dump your .elf as .bin,

Please keep me updated if it all works. Thank you

mastertychy, 2016-11-01:
main.elf is 320244 bytes after my compilation ;)...
I'll compile it again later when i came back to home...

pysco68, 2016-11-01:
Hi again, don't worry for the few bytes unless you're using exactly the same toolchain as mine. Since you're running Ubuntu 14.04LST you're probably using an older version of GCC which might do some other/different optimizations, thus creating a different binary

mastertychy, 2016-11-01:
Hello, 
Now i'm a bit confused...
I've got dev board ZL31ARM (this board have build in SL-LINK v1 compliant programmer). I though that i'll be able to use it for SWD. I connected SWDIO to TMS, SWDCLK to TCK, GND=GND ;)
I was trying to flash uC from Eclipse/OpenOCD - didn't recognize target, probably works JTAG only (?).
ST-Link tool for windows didn't recognize programmer (USB\VID_0483&PID_3744). I'm using non ST drivers (but for that PID 3744 i can't find any other than those from OpenOCD - libusb).
I dont know how to solve the problem...
If everything will failed i'll get orginal StLink/V2 (PID 3748) from my colleague and try...

pysco68, 2016-11-01:
Hi again, sadly I don't own a ST-Link v1 or any compliant programmer but I think that shouldn't be an issue. The connections on the board are definitely SWD (I used a ST-Link v2 clone with the Black Magic Probe firmware on it).

Just a thought: did you power the screen? I mean it must be connected with both USB for power and your programmer.

mastertychy, 2016-11-01:
Yes, i connected it to smartphone charger (2A max), backlight is visible.
I don't know if you were using Eclipse/OpenOCD... when it's connecting with target there is always information that is connected with JTAG.
I'll try on linux again (i had a problem with make stlink.git)...

mastertychy, 2016-11-01:
OK i'm in trouble now ;) can you check it?

kim@kim-ubu:~$ sudo st-flash write /dev/sg0 main.bin 0x8000000
[sudo] password for kim: 
st-flash 1.2.0-144-g9c05837
2016-11-01T21:09:35 INFO /home/kim/stlink.git/src/common.c: Loading device parameters....
2016-11-01T21:09:35 INFO /home/kim/stlink.git/src/common.c: Device connected is: F1 Medium-density device, id 0x13030410
2016-11-01T21:09:35 INFO /home/kim/stlink.git/src/common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x104000 bytes (1040 KiB) in pages of 1024 bytes
2016-11-01T21:09:35 INFO /home/kim/stlink.git/src/sg.c: Successfully opened a stlink v1 debugger
2016-11-01T21:09:35 INFO /home/kim/stlink.git/src/common.c: Attempting to write 7780 (0x1e64) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08001c00 erased
2016-11-01T21:09:35 INFO /home/kim/stlink.git/src/common.c: Finished erasing 8 pages of 1024 (0x400) bytes
2016-11-01T21:09:35 INFO /home/kim/stlink.git/src/common.c: Starting Flash write for VL/F0/F3 core id
2016-11-01T21:09:35 INFO /home/kim/stlink.git/src/flash_loader.c: Successfully loaded flash loader in sram
2016-11-01T21:09:39 ERROR /home/kim/stlink.git/src/flash_loader.c: flash loader run error
2016-11-01T21:09:39 ERROR /home/kim/stlink.git/src/common.c: stlink_flash_loader_run(0x8000000) failed! == -1
stlink_fwrite_flash() == -1

mass erase is working fine...

pysco68, 2016-11-01:
Hi again,

I think you should try to add the "-1" switch as you seem to be using a ST-Link V1

pysco68, 2016-11-01:
Another clue could be the readout protection that might be enabled on your screen; to reset it follow that SO answer: http://stackoverflow.com/questions/32509747/stm32-read-out-protection-via-openocd

pysco68, 2016-11-01:
If you could try both things separately it would be great to understand what's wrong on your side (an easier to advice others for me in the future)

mastertychy, 2016-11-01:
from GDB:

Reading symbols from main.elf...done.
(gdb) target extended-remote :4242
Remote debugging using :4242
0x00000000 in ?? ()
(gdb) monitor swdp_scan
Target does not support this command.
(gdb) attach 1
A program is being debugged already.  Kill it? (y or n) n
Not killed.
(gdb) load
Loading section .text, size 0x1de4 lma 0x8000000
Loading section .data, size 0x80 lma 0x8001de4
Error finishing flash operation
(gdb) attach 1
A program is being debugged already.  Kill it? (y or n) y
Attaching to program: /home/kim/main.elf, Remote target
This target does not support attaching to a process
(gdb) load
Loading section .text, size 0x1de4 lma 0x8000000
Loading section .data, size 0x80 lma 0x8001de4
Error finishing flash operation
(gdb) 

so it's the same... PROG diode is blinking in both GDB console and st-flash (gdb is using st-flash).
I'm not sure if there is memory protection on it. I can and did mass erase.
"-1" option isn't working ;).
OK, i'll try some other methods before i solder uC out ;). 
I'm getting stupid thoughts always when i have a problem... now i'm thinking about soldering out uC and put a new one (i have couple of them), already programmed with JTAG (i'm pretty sure it's still working fine on ST-Link v1, last week i made DC/DC converter on F103RBT6), or solder full JTAG connector directly to uC on LCD board ;).

mastertychy, 2016-11-01:
OK all done ;)
on linux there is no info about memory protection. It's protected even if i can do mass erase!
I did it in fresh windows10 (it was drivers bug on my old machine). 
ST-Link utility told me that there is memory protection enabled.
After disabling everything is OK.

Touch is OK!!!! AMAZING!!! THANKS!!!!

pysco68, 2016-11-02:
Happy you got it working!

So in the end it was the read out protection level still set I guess. I'm going to add that in the readme!
Thank you for your patience!

closed 2016-11-02 by pysco68