pysco68/waveshare-hid #4: waveshare-5inch HDMI LCD (B)-Resistive touchscreen for win10

DRAgon734465502, 2016-11-17:
Hi thank for your program，  I have buy a 5 inch Resistive touchscreen at waveshare

But the 5 inch also have the same problem as the 7inch captive LCD
http://www.waveshare.com/product/modules/oleds-lcds/raspberry-pi-lcd/5inch-hdmi-lcd-b.htm

I think the 5 inch  HDMI LCD (B) have the Similar principles with 7 inch captive LCD

But the 5 inch use stm32 connecting with xtp2046 Not like captive LCD using I2C

Can you give a program about 5 inch  HDMI LCD (B) for win10？？

pysco68, 2016-11-17:
Hello DRAgon734465502,

unfortunately I don't own one of the 5 inch screens, so I can't really help you. Sorry. However you may use my firmware as a starting point for one adapted to your screen. Good luck if you want to go that route!

DRAgon734465502, 2016-11-17:
can you tell me your email？

pysco68, 2016-11-17:
If you want to have a chat, just come over to Gitter

pysco68, 2016-11-17:
Btw. I sent you an invitation to the Gitter room

closed 2016-11-22 by pysco68