pysco68/waveshare-hid #6: Waveshare LCD(C) support

darth-llamah, 2016-11-29:
Waveshare LCD(C)- which has a 1024x600 LCD- works fine with current firmware. All touch events were registered in proper places.
Tested:
-RPi 2: Raspbian, OpenElec
-PC: Windows 10 x64, Debian Stretch

Touchscreen hardware seems not  suited for using higher resolution. After modifying resolution from 800x480 to 1024x600 (for both touch IC initialization and USB descriptor):
-hardware starts and is properly enumerated on the host side
-Y-axis: physical touch on the bottom of the touchscreen is registered by all systems as a touch on the top of the screen, most visible when sliding the touch down
-X-axis- seems to work fine but I didn't test it as much as Y-axis

Once more thank you for your work

pysco68, 2016-11-29:
Hello,

I think I have of clue at what is happening here. If you have a look at https://github.com/pysco68/waveshare-hid/blob/master/src/gt811.c#L513 to https://github.com/pysco68/waveshare-hid/blob/master/src/gt811.c#L517 you'll see that I'm making some calculations there in order to fix the upside down mounted touch digitizer coordinate.

Maybe you could give changing the constant there a try.

darth-llamah, 2016-11-29:
I didn't notice this, thanks.

Also a note about flashing in Windows (I couldn't get my MapleMini based BMP to work under Debian, gdb ends with "volage not implemented" error):
-download Arduino IDE, optionally download a zip and make it portable
-install support for Atmega SAM boards (note: it requires ~300MB)
-gcc toolchain and gdb binaries will be in c:\Users\[user name]\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\bin\ if it's a "normal install" or in case of portable somewhere in [arduino dir]\portable\hardware\

darth-llamah, 2016-11-30:
Ok, gt811.c edited and now touchscreen wors fine using 1024x600 resolution for all tested hosts. However it seems it is not necessary since there were no problems when default 800x480 was in use.
I guess you can add the LCD(C) model to supported hardware.

pysco68, 2016-12-01:
Great to hear. I'll add the information to the readme file when I find some time.

Thank you for the feedback!

closed 2016-12-01 by pysco68