pysco68/waveshare-hid #7: Hanging at I2C

lukas2511, 2016-12-25:
Hey,

I tried to get this running on my display, but I'm seeing a little problem...

My display seems to be identical to yours but has an original STM32 controller instead of the GD32 controller, I replaced everything that seemed to be controller-specific and I got the code to run, but it seems to hang at the following piece of code:

```
    /* Waiting for address is transferred. */
    while (!(I2C_SR1(I2C2) & I2C_SR1_ADDR));    
```

If I disable the gt811 code completely I'm seeing a usb device, so it seems to be running fine except for I2C communication.

After attaching a logic analyzer I can see that it transfers the first I2C packet but stops afterwards, SDA and SCL never seem to change state again, clearing flags and removing the while loop doesn't seem to work either, at the next transmit it just hangs.

I don't have much hope, but do you by any chance have an idea of what may be going wrong? I tried searching for this problem online but I didn't really find much useful information...

pysco68, 2016-12-26:
Hi Lukas,

The STM32 and GD32 behave a tad differently when it comes to I2C. In fact the STM32s I2C peripheral can be called "weird" - and that's a nice term. 

I don't have a device to hand to tell you exactly what to modify; but here's the app note from ST concerning the errata: http://www.st.com/content/ccc/resource/technical/document/application_note/5d/ae/a3/6f/08/69/4e/9b/CD00209826.pdf/files/CD00209826.pdf/jcr:content/translations/en.CD00209826.pdf

Here's a bit of code that know to work:
https://github.com/geoffreymbrown/STM32-Template/blob/master/Library/i2c.c

You'll be able to see that part with 1 byte, 2 bytes and N bytes being transferred, including the irq handling and signal generation stuff going on in between.

pysco68, 2017-01-08:
Hi Lukas,

did you manage anything on this issue?

Btw. what screen is it exactly? It's weird to have the original STM32 on the Chinese' screen. It costs nearly twice as much as the GD32F1 in 1k quantities.

lukas2511, 2017-01-08:
Hey,

I didn't really find the time to experiment some more, but thanks for your reply, will try to get it working soon.

This seems to be a genuine "7inch HDMI LCD (B)" (rev1.1) by Waveshare, got it via Amazon, here are some pictures of the PCB:

[image]
[image]

Also on my board the readout-protect bits weren't set, I got a full firmware dump: [original_firmware.zip](https://github.com/pysco68/waveshare-hid/files/692273/original_firmware.zip)

lukas2511, 2017-01-29:
I tried a few more things (including copying code 1:1 from other repos) but I just can't get the I2C to work, always hangs at sending the address, not sure what's going on...

Since other people seem to have I2C working with the chip maybe it's a problem with my version of libopencm3 or with my compiler or whatever, don't really want to test all that, I already wasted way too many hours trying to get this touchscreen to work...

I'm hoping that maybe somebody else will get this to work and share their solution.

pysco68, 2017-01-29:
Hello Lukas,

sad to hear you didn't find a way through this. I think we should keep the issue open. Eventually somebody will contribute ;)

Thank you for your efforts nonetheless!

ReanimatorRX, 2017-02-17:
Hi
I tried to get this running on my display, but did not work out
[image]
My display seems to be similar  to yours but has an original STM32 controller instead of the GD32

pysco68, 2017-02-20:
Hi @ReanimatorRX on first sight your display looks fairly similar (although I can't see the SWD port anywhere). As I told Lukas, the issue is to be found within the I2C routine(s), as per the errata sheet. If you feel courageous enough, give it a try. There's not much you can do that's really wrong ;) (in the way that if you don't change the IO pin mapping or do some crazy stuff I don't think you can damage any of the chips).

It would be awesome if someone manages to get that running ;)

ReanimatorRX, 2017-02-21:
swd #
[image]

pysco68, 2017-03-10:
Hi @ReanimatorRX did you have any success so far?

robthepyro, 2017-03-24:
@pysco68 @ReanimatorRX 
I have the same board and issue. Close inspection shows the GT811 to be connected to I2C1 and not I2C2. I tried changing the code but no luck (no clock on the SCL pin... ) My first experience with ARM / STM32 so I am still learning about the peripheral setup. 
However the pin mapping is weird! 
It seems 
GT811 INT > PB7 I2C1_SDA
GT811 SDA > PB6 I2C1_SCL (with 10k pullup) 
GT811 SCL > PB5 I2C1_SMBAI (with 10k pullup)  

What is I2C1_SMBAI ?! I can't find much reference... and why this wiring scheme? 

I may try a hardware mod to connect to I2C2 just to check the code is working...

tmclane, 2017-04-27:
Would someone please place the commands necessary to use a STM32 - STLink cheap board to accomplish this flashing?  I like the pricing of the St Link V2 for a one-off flash better than the Blackmagic (especially since they are in pre-order and not available to ship right this second)

ReanimatorRX, 2017-06-03:
Hi
Can you make and send me already compiled file please (bin file)?

pysco68, 2017-06-04:
Hi @ReanimatorRX I'm currently on vacation. I don't have access to my dev machine, so I can't send you anything. Do you have issues getting the stuff compiled on your own?

promzona1, 2017-06-07:
Hi ReanimatorRX  ! You chip STM32F103C8T6 FLASH size 64 Kb. Compiled bin file > 300 Kb.

promzona1, 2017-06-07:
Fake waveshare ?
http://www.waveshare.com/w/upload/6/6d/RPi-LCD-Anti-Piracy-l.jpg

promzona1, 2017-06-08:
To  pysco68:
You are a genius ! ) 
I flashed **v 1.1** (Waveshare 7 LCD (B))
Previously, the sensor did not work in Raspberry Pi. Now works without drivers.
Excellent !
Huge gratitude !

promzona1, 2017-06-08:
How invert x coordinates moves, please ?

ReanimatorRX, 2017-06-08:
Hi promzona1
I flashed main-bin
I'm not seeing a usb device
[main.zip](https://github.com/pysco68/waveshare-hid/files/1061812/main.zip)
my elf file

ReanimatorRX, 2017-06-08:
How did you compile the bin file?

promzona1, 2017-06-13:
arm-none-eabi-objcopy.exe -O binary main.elf main.bin

promzona1, 2017-06-13:
Compiled *.bin from your *.elf :
[main.zip](https://github.com/pysco68/waveshare-hid/files/1070836/main.zip)

promzona1, 2017-06-13:
Hi psyco68 !
How inverted X coordinates in you gt811.c ?
After the flashing, my X coordinates are inverted and the pointer is displayed and moves in a mirror.

_// NOTE: apparently the waveshare touch digitizer was mounted
                // mirrored in both directions, this means we "flip" X <-> Y
                // coordinates AND we have to mirror the Y axis

                // byte[4] X coordinate LSB                
                hid_report->report[4] = data[offsetData + 5];
                
                // byte[5] X coordinate MSB 
                hid_report->report[5] = data[offsetData + 4];

                // Y is inverted for some reason
                // so we have to do some math here...
                // basicall y max - value...
                y_value =  data[offsetData + 3] + (data[offsetData + 2] << 8);
                inverted_y = 480 - y_value;

                // byte[6] Y coordinate LSB
                hid_report->report[6] = inverted_y & 0xFF;

                // byte[7] Y coordinate MSB
                hid_report->report[7] = (inverted_y & 0xFF00) >> 8;_

ReanimatorRX, 2017-06-19:
https://adelectronics.ru/2017/02/01/%D0%BA%D0%BB%D0%BE%D0%BD-%D0%B4%D0%B8%D1%81%D0%BF%D0%BB%D0%B5%D1%8F-7inch-%D0%BE%D1%82-waveshare/

https://adelectronics.ru/2017/05/05/%D0%BF%D1%80%D0%BE%D1%88%D0%B8%D0%B2%D0%BA%D0%B0-%D0%BA%D0%BB%D0%BE%D0%BD%D0%B0-%D0%B4%D0%B8%D1%81%D0%BF%D0%BB%D0%B5%D1%8F-7inch-%D0%BE%D1%82-waveshare/#more-2274

GarageTech, 2021-12-15:
[image]
[image]

Thanks all for help!
I successfully flashed 7inch HDMI LCD (B) 800x480 v 1.1.
And I took a  .bin file from promzona1 and flashed with ST-link V2 under Windows.
The chip on my board is GD32F103.
Then i using it as 1024x600 screen with Raspberry Pi.
/boot/config.txt
disable_overscan=1
overscan_right=225 // these may be different
overscan_bottom=105 // these may be different
framebuffer_width=1024
framebuffer_height=600
hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=87
max_usb_current=1
hdmi_cvt 1024 600 60 0 0 0 0
hdmi_drive=1

At this point you get well adjusted screen but X for the touch was swapped. I not try recompile firmware.
Install edev.

sudo apt-get install xserver-xorg-input-evdev
sudo mv /usr/share/X11/xorg.conf.d/10-evdev.conf /usr/share/X11/xorg.conf.d/45-evdev.conf 

edit /usr/share/X11/xorg.conf.d/45-evdev.conf

Section "InputClass"
        Identifier "evdev touchscreen catchall"
        MatchIsTouchscreen "on"
        MatchDevicePath "/dev/input/event*"
        Driver "evdev"
        Option "InvertX" "true"
EndSection

Reboot.

Now all work fine!!!

simeonking, 2024-09-11:
> [image] [image]
> 
> Thanks all for help! I successfully flashed Waveshare 7inch HDMI LCD (B) 800x480 v1.1. I took .bin file from promzona1 post and flashed with ST-link V2 under Windows. Chip on my board is GD32F103. I using display in 1024x600 mode with RaspberryPi. **/boot/config.txt**
> 
> ```
> disable_overscan=1
> overscan_right=225 // these may be different
> overscan_bottom=105 // these may be different
> framebuffer_width=1024
> framebuffer_height=600
> hdmi_force_hotplug=1
> hdmi_group=2
> hdmi_mode=87
> max_usb_current=1
> hdmi_cvt 1024 600 60 0 0 0 0
> hdmi_drive=1
> ```
> 
> At this point you get well adjusted screen but X for the touch was swapped. I not try recompile firmware. Install evdev.
> 
> sudo apt-get install xserver-xorg-input-evdev sudo mv /usr/share/X11/xorg.conf.d/10-evdev.conf /usr/share/X11/xorg.conf.d/45-evdev.conf
> 
> edit /usr/share/X11/xorg.conf.d/45-evdev.conf
> 
> Section "InputClass" Identifier "evdev touchscreen catchall" MatchIsTouchscreen "on" MatchDevicePath "/dev/input/event*" Driver "evdev" Option "InvertX" "true" EndSection
> 
> Reboot.
> 
> Now all work fine!!!

in my card ,I use x11,/usr/share/X11/xorg.conf.d/10-evdev.conf notwork in my card
so I nano /etc/X11/xorg.conf.d/05-touchscreen.conf 
and used the identity matrix -1 0 1 0 1 0 0 0 1  instead