pysco68/waveshare-hid #8: A false event, when long pressing the button.

Seriga, 2017-01-05:
Windows10IoT ver.14393.576; Rasberry Pi3; 7inch HDMI LCD (C) 1024x600.

On the screen placed 6 buttons. If you press and hold the center button for example, it leads to a chaotic flickering of all the buttons. 
Sometimes the triggering events are “Tapped” buttons which are located nearby.

The effect is amplified if you increase the area of contact, such as tapping with two fingers.

**xaml:**

<Grid Background="{ThemeResource ApplicationPageBackgroundThemeBrush}">
    <GridView x:Name="gridView">
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
    </GridView>
    <TextBlock x:Name="textBlock" TextWrapping="Wrap" Text="" FontSize="24" Margin="10,560,10,10"/>
</Grid>

**cs:**

public sealed partial class MainPage : Page
    {
        public MainPage()
        {
            this.InitializeComponent();
        }

        private void Button_Tapped(object sender, TappedRoutedEventArgs e)
        {
            textBlock.Text = textBlock.Text + " Tapped";
        }
    }

Seriga, 2017-01-05:
xaml:

<Grid Background="{ThemeResource ApplicationPageBackgroundThemeBrush}">
    <GridView x:Name="gridView">
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
    </GridView>
    <TextBlock x:Name="textBlock" TextWrapping="Wrap" Text="" FontSize="24" Margin="10,560,10,10"/>
</Grid>

Seriga, 2017-01-05:
`<Grid Background="{ThemeResource ApplicationPageBackgroundThemeBrush}">
    <GridView x:Name="gridView">
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
    </GridView>
    <TextBlock x:Name="textBlock" TextWrapping="Wrap" Text="" FontSize="24" Margin="10,560,10,10"/>
</Grid>`

Seriga, 2017-01-05:
> <Grid Background="{ThemeResource ApplicationPageBackgroundThemeBrush}">
    <GridView x:Name="gridView">
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
        <Button Height="180" Width="330" Tapped="Button_Tapped" />
    </GridView>
    <TextBlock x:Name="textBlock" TextWrapping="Wrap" Text="" FontSize="24" Margin="10,560,10,10"/>
</Grid>

Seriga, 2017-01-05:
> public sealed partial class MainPage : Page
    {
        public MainPage()
        {
            this.InitializeComponent();
        }

        private void Button_Tapped(object sender, TappedRoutedEventArgs e)
        {
            textBlock.Text = textBlock.Text + " Tapped";
        }
    }

pysco68, 2017-01-08:
Hello Seriga,

I don't own the C model screen and on my B model I didn't manage to reproduce the issue, neither on my app (I have a screen with a good 20 buttons) nor with your code (I just copy-pasted it in a new UWP app and deployed it to the Raspi).

To understand what's happening it would be great to have a log of the HID messages the screen sends. I don't think there's any practical way of finding that out on the raspberry, but if you can connect the screen to a PC and use USBlyzer (http://www.usblyzer.com/ ... even the 30 day trial will do) to trace the HID messages when you "do" the "same gesture" as on your app.

Maybe we manage to track that one down ;)

Just as an aside question: sure your screen is in perfect condition? I had similar experiences with a slightly damaged screen on a phone once. (this is not meant as in "it works on my machine" - developer answer =))

Seriga, 2017-01-08:
Hmm... on Windows 10 x64 everything works as it should. 
Recorded a video of this effect in Windows 10 IoT. 
https://youtu.be/CLdEYUDzY10

pysco68, 2017-01-09:
Hmmm, this is really strange. One last clue maybe; which type of Controller Driver do you have set up?
I'm using the DMMD because the inbox driver gave me perf issues in another regard of my app. (You can switch that in the "Devices" page of the online control panel

Seriga, 2017-01-09:
Changed to DMMD. All the same. Can really had no luck with the display. Or I firmware uploaded “crooked”. In any case, thank you pysco68 for the help.

darth-llamah, 2017-01-09:
Maybe this is related to touch resolution scaling? Windows IoT is a simplified system.
Maybe try flashing this firmware- https://1drv.ms/u/s!AizHkqRak7rko1zpTzGyfs_R7Db-
Only changes made is the reported touch resolution (matching the lcd- 1024x600)

Seriga, 2017-02-03:
Now everything works fine. Set WinIoT 14986

pysco68, 2017-02-04:
Okay, awesome! And thanks for reporting!

closed 2017-02-04 by pysco68