pysco68/waveshare-hid #9: Setting touch sensitivity possible?

ChristianFrisson, 2017-02-07:
Hi,

Great work in reverse-engineering the driver!

To allow retrofitting the screen behind a 2mm-thick plastic protection, would it be possible to modify the threshold or gain of the capacitive sensing so that touch sensing works?

In [gt811.c](https://github.com/pysco68/waveshare-hid/blob/master/src/gt811.c) there may be hints in the GT811 configuration setup:
```
// [53] The screen key threshold
0x50,
    
// [54] Screen loose threshold
0x30,
```

Has anybody ever tried changing these values?

Best regards,
Christian

pysco68, 2017-02-07:
I didn't but it's probably the way to go. You might just give it a try!

In my setup the screen is actually working flawlessly while having a IPad protection screen (those tempered glass ones) applied.

DeadlyJesus, 2017-02-16:
It works !

pysco68, 2017-02-20:
Awesome! Could you share your knowledge? I mean how did you change the threshold and what effect did it have?

Thanks!

DeadlyJesus, 2017-02-21:
Just as Christian said, I changed the "Screen key threshold" and "Screen loose threshold" values to 0x09 and 0x05 respectively. I had to do a bit of tunning (I flashed the screen like 4/5 times) to find theses values, but its what works the best with the 2mm plastic shield I'm using over the touchscreen.
Before I could not detect any touch while the shield was over the screen, with those settings I can now use it almost as if the shield was not there
Huge thanks to you.

ChristianFrisson, 2017-02-21:
Good job @DeadlyJesus !

(Full disclosure: @DeadlyJesus and I are currently sharing the same office)

closed 2017-02-21 by ChristianFrisson

pysco68, 2017-02-21:
@DeadlyJesus you're welcome! Thank you for getting back here. I think this will be useful for others!