pysco68/waveshare-hid #10: Touch to screen without USB connection cause  firmware failure.

fa1ke5, 2017-12-26:
If im touch to screen on power on time, i get descriptor failure in OS and i need to reconnect display to USB (or reset controller )onesmore.
Im simulate this trouble on gdb session, connect display to standalone power source and start gdb

GNU gdb (7.10-1ubuntu3+9) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from main.elf...done.
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) monitor swdp_scan
Target voltage: unknown
Available Targets:
No. Att Driver
 1      STM32F1 medium density
(gdb) att 1
Attaching to program: /home/madmax/stm32/waveshare-hid/src/main.elf, Remote target
0x08001bbe in st_usbfs_poll (dev=0x20000180 <st_usbfs_dev>) at ../common/st_usbfs_core.c:258
258		if (istr & USB_ISTR_WKUP) {
(gdb) load
Loading section .text, size 0x1dc8 lma 0x8000000
Loading section .data, size 0x80 lma 0x8001dc8
Start address 0x80015a4, load size 7752
Transfer rate: 7 KB/sec, 861 bytes/write.
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/madmax/stm32/waveshare-hid/src/main.elf 

Program received signal SIGSEGV, Segmentation fault.
0x08001562 in st_usbfs_copy_to_pm (vPM=<optimized out>, buf=<optimized out>, len=5, len@entry=11) at ../st_usbfs_v1.c:61
61		for (len = (len + 1) >> 1; len; len--) {
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/madmax/stm32/waveshare-hid/src/main.elf 

// In this time im touch to screen one time and wuola.... Segmentation fault.

Program received signal SIGSEGV, Segmentation fault.
0x08001562 in st_usbfs_copy_to_pm (vPM=<optimized out>, buf=<optimized out>, len=5, len@entry=11) at ../st_usbfs_v1.c:61
61		for (len = (len + 1) >> 1; len; len--) {
(gdb) 

********
In active USB coonection case touch screen work nice.

pysco68, 2018-01-01:
Hi @fa1ke5,

nice catch! I guess this is an issue with `send_hid_report()` (in https://github.com/pysco68/waveshare-hid/blob/e0784af33ed352d2ac72ba3511d1f3d44a771252/src/usb_device.c#L248) not checking if the  USB communication is correctly set up. 

I won't be back to my lab for a few more days, so I don't have access to a screen to try and fix this issue.
If you want you can give it a try, I'll happily accept a PR ;)

Have a nice days

fa1ke5, 2018-01-09:
Hi @pysco68, i dont have aditional expirence in embedded C++ for modifing source code correctly.
But i have a lot of display's and may testing code on them's.
I think we are no need checking setup USB all time. May be also initialise GT811 touchscreen controller and recieve data from him only after USB recieve host signal?

fa1ke5, 2018-04-02:
Hi all!
i'm solve software crash and add some tuning
1. in usb_device.c 
    while(usbd_ep_write_packet(usbd_dev, ENDPOINT_ADDRESS, hid_report, len) != len);
i change to
    while(usbd_ep_write_packet(usbd_dev, ENDPOINT_ADDRESS, hid_report, len) != 0);
and fix seg. fault, code is working, but still freeze touch if i'm touch to screen continusly in power on PC procedure.
2. Im adding in usb_device.c varable usbOK

#include <libopencm3/usb/usbd.h>
#include <libopencm3/usb/hid.h>
#include <stdlib.h>
#include "usb_device.h"
#include "systick.h"
**int usbOK = 0;**

and

void hid_set_config(usbd_device *cb_usbd_dev, uint16_t wValue)
{
    (void)wValue;
    (void)cb_usbd_dev;

    usbd_ep_setup(cb_usbd_dev, ENDPOINT_ADDRESS, USB_ENDPOINT_ATTR_INTERRUPT, 4, NULL);

    usbd_register_control_callback(
                cb_usbd_dev,                                        // usb device
                USB_REQ_TYPE_STANDARD | USB_REQ_TYPE_INTERFACE,     // request type
                USB_REQ_TYPE_VENDOR | USB_REQ_TYPE_RECIPIENT,       // request mask
                hid_control_request);                               // callback
msleep(1000);
**usbOK=1;**
}

in main.c
        while (1) {
        poll_usb();
**if(usbOK == 1)**{
        gt811_poll();
                }
        }
}

in gt811.c
// send that report!
            send_hid_report(hid_report->report, 11);

            **msleep(1);**
            // remember we sent a report
            sentReports++;

in usb_dev.h

void setup_usb(void);
void poll_usb(void);
void hid_set_config(usbd_device *cb_usbd_dev, uint16_t wValue);
void send_hid_report(uint8_t *hid_report, uint8_t len);
int hid_control_request(usbd_device *cb_usbd_dev, struct usb_setup_data *req, uint8_t **buf, uint16_t *len, usbd_control_complete_callback *complete);
**extern int usbOK;**

And all work pretty good.

And now, im change the VID i PID from usb_device.c and change rosolution touch screen for get 1024x600
//    0x26, 0x20, 0x03,                   //       LOGICAL_MAXIMUM (800)
    **0x26, 0x00, 0x04,                   //       LOGICAL_MAXIMUM (1024)**
    0x35, 0x00,                         //       PHYSICAL_MINIMUM (0)
    0x46, 0xFF, 0xFF,                   //       PHYSICAL_MAXIMUM (x)
    0x09, 0x30,                         //       USAGE (X)
    0x81, 0x02,                         //       INPUT (Data,Var,Abs)
    0x46, 0xFF, 0xFF,                   //       PHYSICAL_MAXIMUM (y)
//    0x26, 0xe0, 0x01,                   //       LOGICAL_MAXIMUM (480)
    **0x26, 0x58, 0x02,                   //       LOGICAL_MAXIMUM (600)**

//    .idVendor = 0x6666,         // prototype VID
//    .idProduct = 0xBEEF,        // dummy
    **.idVendor = 0x222a,         // prototype VID
    .idProduct = 0x0001,        // dummy**

and in gt811.c

//    0xe0,0x01,  // 480
    **0x58,0x02,  // 600**

    // [61 - 62] The Y coordinate outputs the maximum value
//    0x20,0x03,  // 800
    **0x00,0x04,  // 1024**

pysco68, 2018-04-03:
Hi @fa1ke5 thank you for the contributions, I'll have a closer look at them asap (busy moving my lab atm). I guess that I'll have to try some other edge cases (like host resets without power loss) to eventually reset that `usbOk` flag to remain safe under all conditions.

Note about your idea to increase the resolution of the touch screen:
I tried this too during development but read *somewhere* that earlier versions of Android and Windows (I think it was 7 and earlier) don't always handle differences in resolution of the touch screen digitizer properly (especially native screen resolution < digitizer resolution). I figured that the physical resolution of my finger on the screen would be way lower anyway.

But don't quote me on that, it's just dark memories ;)

fa1ke5, 2018-04-03:
i have 1024x600 display and need stylus anytime, 1024x600 touch work better, but windows driver seeing this resolution correctly only with alternate VID and PID.
Im using 222a and 0001.