pysco68/waveshare-hid #12: My board has a genuine STM32 (I think) and is bricked after upload

kargeor, 2020-01-19:
Do I need to change anything to build for STM32 instead of GD32?

pysco68, 2020-02-04:
Hi Kargeor,

I'm sorry to hear you've got trouble there. I never got my hands on a STM32 based screen, so I can't tell you what the excat difference(s) is/are. 

Just browsing the sources quickly, there seem to be a few GD32-specific includes (mostly regarding the system clock configuration and I2C) but I think these things were designed to be identical the the STM32 (still this is of the top of my head, so might be worth some investigation), so _/\O/\_

OTOH, I can't guarantee that you screen doesn't have another hardware change vs. what I got my hands on years ago. 

Could you post some high-res pictures here? 

Regards,
Yannic

kargeor, 2020-02-15:
Hey,
I setup GT811 to talk directly to my raspberry pi using I2C. 
Here is the code: https://github.com/kargeor/gt811-virtual_touchscreen

I don't plan to go back to USB but I can share a photo of the board:
[image]